zephyrproject-rtos/meta-zephyr-sdk #91: Zephyr's arm-zephyr-eabi-as reports "Error: unaligned opcodes detected in executable segment" when building code for frdm_kw41z

pfalcon, 2018-01-15:
[ecma-helpers-string.s.txt](https://github.com/zephyrproject-rtos/meta-zephyr-sdk/files/1631606/ecma-helpers-string.s.txt)
There's an error When building JerryScript Zephyr port for BOARD=frdm_kw41z, originally hit by Linaro CI build: https://ci.linaro.org/view/lite-iot-ci/job/lite-aeolus-jerryscript/202/PLATFORM=frdm_kw41z,ZEPHYR_GCC_VARIANT=zephyr,label=docker-xenial-amd64-13/console . Relevant:

~~~
02:00:54 /tmp/cck9bMW2.s: Assembler messages:
02:00:54 /tmp/cck9bMW2.s: Error: unaligned opcodes detected in executable segment
~~~

As can be seen, error reporting is stellar: no line number, no nothing. Reproducing the error locally, it happens with the following command line:

~~~
/home/pfalcon/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc -DCONFIG_DISABLE_BUILTINS -DCONFIG_DISABLE_ES2015 -DCONFIG_DISABLE_UNICODE_CASE_CONVERSION -DCONFIG_MEM_HEAP_AREA_SIZE=16384 -DJERRY_ENABLE_ERROR_MESSAGES -DJERRY_NDEBUG -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/api -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/debugger -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/ecma/base -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/ecma/builtin-objects -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/ecma/builtin-objects/typedarray -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/ecma/operations -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/jcontext -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/jmem -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/jrt -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/lit -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/parser/js -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/parser/regexp -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/vm -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-libm/include  -fno-builtin -fno-stack-protector -Wall -Werror=all -Wextra -Werror=extra -Wformat-nonliteral -Werror=format-nonliteral -Winit-self -Werror=init-self -Wconversion -Werror=conversion -Wsign-conversion -Werror=sign-conversion -Wformat-security -Werror=format-security -Wmissing-declarations -Werror=missing-declarations -Wshadow -Werror=shadow -Wstrict-prototypes -Werror=strict-prototypes -Wundef -Werror=undef -Wold-style-definition -Werror=old-style-definition -Wno-stack-protector -Wno-attributes -Wlogical-op -Werror=logical-op -std=c99 -pedantic -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-stack-protector -fno-strict-overflow -ffreestanding -fno-reorder-functions -fno-defer-pop -fdata-sections -ffunction-sections -fno-inline-functions -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DCPU_MKW41Z512VHT4 -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/build/frdm_kw41z/zephyr/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -std=c99 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -ffunction-sections -fdata-sections -mabi=aapcs --sysroot /home/pfalcon/opt/zephyr-sdk-0.9.2/sysroots/armv5-zephyr-eabi/usr -mthumb -mcpu=cortex-m0plus -isystem/mnt/hdd/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include -isystem/mnt/hdd/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include-fixed -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/kernel/include -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/arch/arm/include -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/arch/arm/soc/nxp_kinetis/kwx -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/arch/arm/soc/nxp_kinetis/kwx/include -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/arch/arm/soc/nxp_kinetis/include -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/boards/arm/frdm_kw41z -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/drivers -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/build/frdm_kw41z/zephyr/zephyr/include/generated -isystem/mnt/hdd/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include -isystem/mnt/hdd/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/include-fixed -isystem/home/pfalcon/opt/zephyr-sdk-0.9.2/sysroots/armv5-zephyr-eabi/usr/include -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/ext/hal/cmsis/Include -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/ext/hal/nxp/mcux/devices/MKW41Z4 -isystem/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/ext/hal/nxp/mcux/drivers/. -Wall -Wno-format-zero-length -Wno-pointer-sign -Werror=format -Werror=implicit-int -Wno-unused-but-set-variable -Wno-main -Wno-strict-aliasing -Wno-old-style-declaration -Wno-error=format= -D_XOPEN_SOURCE=700 -Wno-error=conversion -Os -DNDEBUG   -o CMakeFiles/jerry-core.dir/ecma/base/ecma-helpers-string.c.obj   -c /home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c
~~~

Offending .s file, generating by adding -S to the command above, is attached.

pfalcon, 2018-01-15:
@MaureenHelm, @dleach02: I'm trying to dig into this myself, but if you have any suggestions, they're appreciated.

pfalcon, 2018-01-15:
So, frdm_kw41z is Cortex-M0. Per @carlescufi also tried BOARD=nrf51_pca10028, and it leads to the same error. So, it's a Cortex-M0 codegeneration problem.

pfalcon, 2018-01-15:
@galak : FYI

nashif, 2018-01-15:
did you try  to build with another compiler?

pfalcon, 2018-01-15:
@nashif : I didn't. I can try, but the aim is anyway to build with Zephyr toolchain.

nashif, 2018-01-15:
@pfalcon 
> but the aim is anyway to build with Zephyr toolchain.

yeah, but it help to know if this is an issue in the Zephyr SDK or something general

pfalcon, 2018-01-15:
Will do. In the meantime:

~~~
$ /home/pfalcon/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-as --version
GNU assembler (GNU Binutils) 2.27.0.20160806
~~~

pfalcon, 2018-01-15:
With ARM GCC https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/6-2017q2/gcc-arm-none-eabi-6-2017-q2-update-linux.tar.bz2 (tried this first because that's what we use in Linaro CI):

~~~
$ ./arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

$ ./arm-none-eabi-as --version
GNU assembler (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 2.28.0.20170620
~~~

The same error:

~~~
cd /home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/build/frdm_kw41z/obj/jerry-core && /home/pfalcon/toolchain/gcc-arm-none-eabi-6-2017-q2-update/bin/arm-none-eabi-gcc -DCONFIG_DISABLE_BUILTINS [...]
[...]
/tmp/ccpJ7vZd.s: Assembler messages:
/tmp/ccpJ7vZd.s: Error: unaligned opcodes detected in executable segment
jerry-core/CMakeFiles/jerry-core.dir/build.make:350: recipe for target 'jerry-core/CMakeFiles/jerry-core.dir/ecma/base/ecma-helpers-string.c.obj' failed
~~~

pfalcon, 2018-01-15:
Next specimen: gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2 from https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads , release 2017-12-18:

~~~
$ ./arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]

$ ./arm-none-eabi-as --version
GNU assembler (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 2.29.51.20171128
~~~

Here, finally success at building:

~~~
make[1]: Leaving directory '/mnt/hdd/projects-3rdparty/Embedded/Zephyr/jerryscript/build/frdm_kw41z/zephyr'
Finished
   text	   data	    bss	    dec	    hex	filename
 130540	   2740	  23872	 157152	  265e0	build/frdm_kw41z/zephyr/zephyr/zephyr.elf
~~~

Will check if it runs shortly.

pfalcon, 2018-01-15:
> Will check if it runs shortly.

It does.

nashif, 2019-05-01:
outdated

closed 2019-05-01 by nashif