zephyrproject-rtos/meta-zephyr-sdk #92: Upgrade dtc to V1.4.2

erwango, 2018-03-28:
In order to enable definitions of connectors in device tree, and allow combination of boards and expansion boards without knowing a priori which connector will be used, an evolution of dtc is required.
It has been implemented in dtc v4.1.2:
dtc: parser: Add label while overriding nodes [1]

[1]: https://git.kernel.org/pub/scm/utils/dtc/dtc.git/commit/?h=v1.4.6&id=3346e065aacb1f05bdfb3872e3ba709b5f895846

nashif, 2018-04-22:
try this:

https://github.com/zephyrproject-rtos/meta-zephyr-sdk/releases/tag/0.9.3-rc2

erwango, 2018-04-23:
@nashif : Working, thanks!
Though, now I have to solve https://github.com/zephyrproject-rtos/zephyr/issues/7013,
which are warning introduced in between. Working on it

nashif, 2018-05-11:
fixed

closed 2018-05-11 by nashif