zephyrproject-rtos/meta-zephyr-sdk #112: install script SDK 0.9.5 seems not working

chrislynn, 2018-11-30:


chrislynn, 2018-11-30:
Following setup step to run the .run script is giving back error. Try with older release SDK .run script 0.9.4 works fine.

pfalcon, 2018-11-30:
Which error? Zephyr master requires 0.9.5, doesn't work with any older version, so *a lot* of people installed SDK 0.9.5 without problems, including e.g. me.

chrislynn, 2018-11-30:
sh zephyr-sdk-0.9.5-setup.run
zephyr-sdk-0.9.5-setup.run: 8: zephyr-sdk-0.9.5-setup.run: Syntax error: newline unexpected

pfalcon, 2018-11-30:
Do you have a file of size 301MB, as appears at https://github.com/zephyrproject-rtos/meta-zephyr-sdk/releases ? Re-download if in doubt. @nashif: Worth publishing md5/sha1.

Otherwise, which system do you use? Try `bash zephyr-sdk-0.9.5-setup.run`.

chrislynn, 2018-11-30:
Hi,
Thanks for the advice. Redownload and it works fine this time. Not sure what is problem with previous file.
By the way, I came to python dependencies installation error now. Can you give some light? I am not following the suggestion of Ubuntu 16.04 as I only have a Ubuntu 14 system.
gchee2@gchee2-desk01:/media/1TB_HDD/zephyr_sdk$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 14.04.5 LTS
Release:        14.04
Codename:       trusty
gchee2@gchee2-desk01:/media/1TB_HDD/zephyr_sdk$

[cid:image002.jpg@01D488BE.23D34880]

nashif, 2019-05-01:
Moved to a new SDK architecture, please re-open if still exists.

closed 2019-05-01 by nashif