zephyrproject-rtos/meta-zephyr-sdk #113: Could not flash nucleo_f746zg

erwango, 2018-12-08:
**Describe the bug**
It's not possible to flash nucleo_f746zg board

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=board\nucleo_f746zg
3. ninja flash
4. See error

```
/local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32f7x.cfg:155: Error: invalid command name "stm32f7x.dap"
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "/local/mcu/zephyr/zephyr-project/boards/arm/nucleo_f746zg/support/openocd.cfg", line 1
at file "/local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/board/st_nucleo_f7.cfg", line 8
at file "/local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32f7x.cfg", line 155
ERROR: command exited with status 1: /local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /local/mcu/zephyr/zephyr-project/boards/arm/nucleo_f746zg/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /local/mcu/zephyr/zephyr-project/samples/drivers/counter_rtc/outdir/nucleo_f746zg/zephyr/zephyr.bin 0x8000000' -c 'reset halt' -c 'verify_image /local/mcu/zephyr/zephyr-project/samples/drivers/counter_rtc/outdir/nucleo_f746zg/zephyr/zephyr.bin 0x8000000' -c 'reset run' -c shutdown
```

**Expected behavior**
Should flash.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
```
/local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32f7x.cfg:155: Error: invalid command name "stm32f7x.dap"
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "/local/mcu/zephyr/zephyr-project/boards/arm/nucleo_f746zg/support/openocd.cfg", line 1
at file "/local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/board/st_nucleo_f7.cfg", line 8
at file "/local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32f7x.cfg", line 155
ERROR: command exited with status 1: /local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /local/mcu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /local/mcu/zephyr/zephyr-project/boards/arm/nucleo_f746zg/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /local/mcu/zephyr/zephyr-project/samples/drivers/counter_rtc/outdir/nucleo_f746zg/zephyr/zephyr.bin 0x8000000' -c 'reset halt' -c 'verify_image /local/mcu/zephyr/zephyr-project/samples/drivers/counter_rtc/outdir/nucleo_f746zg/zephyr/zephyr.bin 0x8000000' -c 'reset run' -c shutdown
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA: 14084e99608bf6ce31a14c77219ea0047d2a88c9

**Additional context**
Workaround is to drag and drop .bin file in mounted disk

erwango, 2019-01-11:
@inakypg, this should solve the issue you met with same board

nashif, 2019-05-01:
fixed in latest SDK

closed 2019-05-01 by nashif