zephyrproject-rtos/meta-zephyr-sdk #114: arc emsk's openocd debugger doesn't work in Zephyr SDK 0.9.5

vonhust, 2018-12-08:
**Describe the bug**
After Zephyr SDK updated to 0.9.5,  the openocd debugger of arc em_starterkit doesn't work correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=em_starterkit_em7d ..
3. make
4. west debug
5. see error

**Expected behavior**
the debugger cann't run because of error caused by openocd.

**Impact**
can't debug the em_starterkit hardware

**Screenshots or console output**
The log is as follow:

Using runner: em-starterkit
Open On-Chip Debugger 0.10.0+dev-gea2753a7-dirty (2018-11-02-23:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Info : JTAG tap: arc-em.cpu tap/device found: 0x200044b1 (mfg: 0x258 (ARC International), part: 0x0004, ver: 0x2)
**Register `identity' is not found.**
in procedure 'init' 
in procedure 'ocd_bouncer'

Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* arc-em.cpu         arcv2      little arc-em.cpu         halted
Info : Listening on port 6333 for tcl connections
Info : Listening on port 4444 for telnet connections
GNU gdb (GDB) 7.12.0.20161010-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pokysdk-linux --target=arc-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/wren/zephyr/samples/philosophers/emsk/zephyr/zephyr.elf...done.
Info : accepting 'gdb' connection on tcp/3333
Remote debugging using :3333
warning: while parsing target description (at line 6): Boolean fields must be one bit in size
**warning: Could not load XML target description; ignoring
openocd: src/server/gdb_server.c:1188: gdb_get_registers_packet: Assertion `reg_packet_size > 0' failed.
Remote connection closed**

**Environment (please complete the following information):**
 - OS: Linux/Debian
 - Toolchain (Zephyr SDK)
 - latest commit

**Additional context**
Add any other context about the problem here.

abrodkin, 2019-04-25:
@ruuddw are you referring to this change https://github.com/ntfreak/openocd/commit/2231da8ec4e7d7ae9b652f3dd1a7104f5a110f3f?

ruuddw, 2019-04-25:
Yes, that change seems to introduce this 'dap_init'.

abrodkin, 2019-04-26:
So the problem has nothing to do with DAP, everything is much simpler.
There's an interesting change introduced in https://github.com/zephyrproject-rtos/openocd/commit/b5964191f0d2fc3ace607af001df3d57cbfbaf2b (basically that's an upstream fix, see https://repo.or.cz/openocd.git/commit/b5964191f0d2fc3ace607af001df3d57cbfbaf2b).

What it does it skips all the registers that don't have `exist = true` in `register_get_by_name()`.
But in `arc_set_reg_exists()` we use exactly `register_get_by_name()` to set some registers as **existing** and we essentially fail as `register_get_by_name()` skips those **not existing** :)

That said the simplest fix is removal of 1 hunk of mentioned patch:
```
diff --git a/src/target/register.c b/src/target/register.c
index 850641448..a6e29ac3f 100644
--- a/src/target/register.c
+++ b/src/target/register.c
@@ -44,8 +44,6 @@ struct reg *register_get_by_name(struct reg_cache *first,

        while (cache) {
                for (i = 0; i < cache->num_regs; i++) {
-                       if (cache->reg_list[i].exist == false)
-                               continue;
                        if (strcmp(cache->reg_list[i].name, name) == 0)
                                return &(cache->reg_list[i]);
                }
```
but it might re-introduce a problem mentioned patch was supposed to fix though from function name it's not clear what is a problem with returning a pointer to a register which is described but doesn't exist (for any reason). 

So maybe have something like that:
```
diff --git a/src/target/register.c b/src/target/register.c
index 850641448..3b5f22377 100644
--- a/src/target/register.c
+++ b/src/target/register.c
@@ -44,7 +44,7 @@ struct reg *register_get_by_name(struct reg_cache *first,

        while (cache) {
                for (i = 0; i < cache->num_regs; i++) {
-                       if (cache->reg_list[i].exist == false)
+                       if (cache->reg_list[i].exist == false && !search_all)
                                continue;
                        if (strcmp(cache->reg_list[i].name, name) == 0)
                                return &(cache->reg_list[i]);
```
@sstallion could you please comment on this one?
@nashif what do you think, may we accept such a hack for now?

sstallion, 2019-04-26:
@abrodkin Interesting. Targets don't typically use `register_get_by_name` when setting up registers; normally this is done when building the register cache as a part of examine. When it comes to register numbering, OpenOCD is at the mercy of GDB; if the underlying GDB implementation skips around (eg. eSi-RISC targets) you have a situation where not all registers exist when requested. Looking at the proposed patch, I wouldn't go down this path as it would likely re-introduce the original bug that was fixed. I'm curious why you're calling into `register_get_by_name` - RTOS support typically relies on the `rtos_register_stacking` structs, which will contain the necessary information about a given register (eg. number/offset/width).

sstallion, 2019-04-26:
BTW, I'm happy to help out if you need some OpenOCD expertise to get this resolved.

abrodkin, 2019-04-26:
@sstallion this particular problem has nothing to do with RTOS I guess as we're talking about very basic target initialization. Which I would think is the same for any type of binary we're going to debug. Or maybe that's just my lack of OpenOCD knowledge - then please pardon myself being so naive.

Then how can we solve an interesting problem?

Depending on CPU configuration we may have features X, Y, Z. For each feature we have a set of registers (i.e. if feature X is not configured its registers won't exist**). What's worse on attempt to read that not-implemented register CPU raises "Instruction error" exception.

Fortunately for each feature there's a special Build Configuration Register (BCR) which encodes feature version and some more details. So we first read contents of all BCRs (or those we care about) and then depending on what we get we mark certain set of registers as **existing**. And the simplest solution was [given all registers so far have unique names] to get a pointer to register's structure with help of `register_get_by_name ()` and then set **exist** flag.

sstallion, 2019-04-26:
Gotcha. This sounds very similar to eSi-RISC. If you could point me to the
sources for your target support I could take a quick look and see if
anything comes to mind.

abrodkin, 2019-04-26:
@sstallion Origin is here https://github.com/foss-for-synopsys-dwc-arc-processors/openocd/ - OpenOCD fork for ARC cores, though Zephyr uses re-based version available here - https://github.com/zephyrproject-rtos/openocd/.

ARC-related changes are pretty much everywhere - in `src/arget`, `tcl` etc.
Maybe https://github.com/foss-for-synopsys-dwc-arc-processors/openocd is a better source as all last commits have to do with ARC stuff so it's easier to see what is what.
For example the board we discussing here is:
 - https://github.com/foss-for-synopsys-dwc-arc-processors/openocd/blob/master/tcl/board/snps_em_sk_v2.2.cfg
 - https://github.com/foss-for-synopsys-dwc-arc-processors/openocd/blob/master/tcl/target/snps_em_sk_fpga.cfg
 - https://github.com/foss-for-synopsys-dwc-arc-processors/openocd/blob/master/tcl/cpu/arc/em.tcl
 - https://github.com/foss-for-synopsys-dwc-arc-processors/openocd/tree/master/src/target/arc_*

I'll be happy to answer any questions regarding ARC stuff.

sstallion, 2019-04-26:
Thanks! I should be able to get some eyes on it this weekend. I'm hoping
all that's needed is a minor change in how the register cache is built.
I'll let you know if I find anything.

sstallion, 2019-04-27:
@abrodkin I took a quick look at the target support. Everything looks as though it is behaving correctly at initialization and after an examine. It looks to me that the core issue is that having `arc_set_reg_exists` call into `register_get_by_name` is the wrong approach. I would suggest not using this function to lookup non-existent registers. This is a very short function, I would suggest duplicating the necessary bits in `arc_set_reg_exists` instead.