zephyrproject-rtos/meta-zephyr-sdk #116: broken link for new SDK

andrewboie, 2019-03-20:
I can't build due to this error now:
```

$ cmake ..
Zephyr version: 1.14.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.2", minimum required is "3.4") 
-- Selected BOARD nsim_sem
CMake Error at /projects/zephyr2/zephyr/cmake/toolchain/zephyr/host-tools.cmake:36 (message):
  The SDK version you are using is too old, please update your SDK.

  You need at least SDK version 0.10.0.

  You have version 0.9.5 (/opt/zephyr-sdk).

  The new version of the SDK can be downloaded from:

  
  https://github.com/zephyrproject-rtos/meta-zephyr-sdk/releases/download/0.10.0/zephyr-sdk-0.10.0-setup.run

Call Stack (most recent call first):
  /projects/zephyr2/zephyr/cmake/host-tools.cmake:1 (include)
  /projects/zephyr2/zephyr/cmake/app/boilerplate.cmake:379 (include)
  CMakeLists.txt:2 (include)

-- Configuring incomplete, errors occurred!
```

But the provided link is broken

https://github.com/zephyrproject-rtos/meta-zephyr-sdk/releases/download/0.10.0/zephyr-sdk-0.10.0-setup.run

b0661, 2019-03-31:
Wrong path in error message - should be:

wget https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.10.0/zephyr-sdk-0.10.0-setup.run

nashif, 2019-05-01:
fixed in PR https://github.com/zephyrproject-rtos/zephyr/pull/14778

closed 2019-05-01 by nashif